TinkerMill/tinkerAccess #17: Unlock Time Incompatible with Door Access Solenoids

RonaldThomas, 2018-12-07:
1) The smallest unit of time the software considers for granting access is one minute and I'm pretty sure it's stored as an integer. We want to be able to specify time in seconds.
2) IIRC, the refactor of the client software began retrieving the access time granted for each badge in event from the deviceAccess table in the SQL database on the server. The records in deviceAccess are unique combinations of users and devices which are created when a trainer gives someone access to a device. Currently, the server software automatically populates the time field for each record with "100" when the record is created. If we want the access time to be specified for each user in deviceAccess, we should change server.py to pull in the time value as a variable from somewhere instead of as a hard-coded value.

ToddFranks-TM, 2020-12-10:
PR #44 addresses this issue. Instead of modifying the server side to send the time value, this is handled as a special "door" parameter on the client to set the access time and ignore the server value sent.

ToddFranks-TM, 2021-09-21:
This issue has been addressed, so I am closing

closed 2021-09-21 by ToddFranks-TM