TinkerMill/tinkerAccess #26: No Exception Handling Around I2C Communications

RonaldThomas, 2020-05-30:
With some frequency, the TinkerAccess node that we have installed on the Laser Cutter seems to be crashing and a full hardware restart is required to recover. Checking syslog on that device shows errors like these:

`pi@lasercutterpi:~ $ cat /var/log/syslog.1 | grep Errno
May 20 12:08:58 lasercutterpi [Errno 121] Remote I/O error tinker-access-client (2017.07.19.18) 4 Laser Cutter - RDWorks 210 Ron Thomas 150060E726B4
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 242, in write self.__write_to_lcd(first_line, second_line)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 129, in __write_to_lcd lcd.write(first_line, second_line)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 179, in write self.lcd_display_string(first_line, 1, 0)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 156, in lcd_display_string self.lcd_write(ord(char), Rs)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 122, in lcd_write self.lcd_write_four_bits(mode | ((cmd << 4) & 0xF0))
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 126, in lcd_write_four_bits self.lcd_device.write_cmd(data | LCD_BACKLIGHT)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/I2CApi.py", line 46, in write_cmd self.bus.write_byte_data(self.addr, cmd, 0)
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 204, in write_byte_data ioctl(self.fd, I2C_SMBUS, msg)
IOError: [Errno 121] Remote I/O error
May 20 12:09:15 lasercutterpi [Errno 121] Remote I/O error tinker-access-client (2017.07.19.18) 4 Laser Cutter - RDWorks 210 Ron Thomas 150060E726B4
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 242, in write self.__write_to_lcd(first_line, second_line)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 128, in __write_to_lcd with LcdApi() as lcd:
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 102, in __init__ self.__init__display()
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 113, in __init__display self.lcd_write(LCD_FUNCTIONSET | LCD_2LINE | LCD_5x8DOTS | LCD_4BITMODE)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 122, in lcd_write self.lcd_write_four_bits(mode | ((cmd << 4) & 0xF0))
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 126, in lcd_write_four_bits self.lcd_device.write_cmd(data | LCD_BACKLIGHT)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/I2CApi.py", line 46, in write_cmd self.bus.write_byte_data(self.addr, cmd, 0)
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 204, in write_byte_data ioctl(self.fd, I2C_SMBUS, msg)
IOError: [Errno 121] Remote I/O error
May 20 12:09:29 lasercutterpi [Errno 121] Remote I/O error tinker-access-client (2017.07.19.18) 4    
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 242, in write self.__write_to_lcd(first_line, second_line)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 129, in __write_to_lcd lcd.write(first_line, second_line)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 179, in write self.lcd_display_string(first_line, 1, 0)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 156, in lcd_display_string self.lcd_write(ord(char), Rs)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 122, in lcd_write self.lcd_write_four_bits(mode | ((cmd << 4) & 0xF0))
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 127, in lcd_write_four_bits self.lcd_strobe(data)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/LcdApi.py", line 131, in lcd_strobe self.lcd_device.write_cmd(data | En | LCD_BACKLIGHT)
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/I2CApi.py", line 46, in write_cmd self.bus.write_byte_data(self.addr, cmd, 0)
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 204, in write_byte_data ioctl(self.fd, I2C_SMBUS, msg)
IOError: [Errno 121] Remote I/O error
May 20 12:09:30 lasercutterpi [Errno 121] Remote I/O error tinker-access-client (2017.07.19.18) 4    
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/Client.py", line 593, in run client.wait()
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/Client.py", line 500, in wait self.__device.wait()
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 261, in wait self.__raise_fault()
  File "/usr/local/lib/python2.7/dist-packages/tinker_access_client/DeviceApi.py", line 265, in __raise_fault raise self.__fault
IOError: [Errno 121] Remote I/O error`

While we haven't yet identified the ultimate root cause of the problem, it would be great to add exception handling to the relevant sections of the code so it could least make an attempt at self-recovery.

lucosmic, 2020-06-11:
I don't know if this is to be marked as a separate issue, but same thing happened when it couldn't connect to the TinkerMill WiFi, if I'm not mistaken.

RonaldThomas, 2020-06-11:
> I don't know if this is to be marked as a separate issue, but similar thing happened when it couldn't connect to the TinkerMill WiFi, if I'm not mistaken.

The WiFi connectivity problems are a separate issue and (we believe) unrelated to the TinkerAccess code.

lucosmic, 2020-06-11:
Yes connectivity problems may be unrelated to TinkerAccess, but I remember the TinkerAccess node not explicitly telling me on the LCD panel that it couldn't connect to the Wi-Fi, for users to help diagnose the problem. Anyway, I'm familiarizing myself with the code.

lttech-llc, 2020-06-27:
The new release v2020.06.27.23 has fixes for this issue. It captures any I2C bus errors and will retry 5 times before declaring a fatal error, forcing a reboot.

closed 2020-06-27 by RonaldThomas