TinkerMill/tinkerAccess #34: Screw Terminals Too Tall

RonaldThomas, 2020-07-12:
The PCB designs so far have accommodated some screw terminals TinkerMill has happened to have on hand. Unfortunately, these terminals are rather tall and protrude higher than the top surface of the RFID module. It would be great if future PCB designs used something like ASSMAN WSW A-TB350TE02 ([DigiKey A-TB350-TE02-ND](https://www.digikey.com/product-detail/en/assmann-wsw-components/A-TB350-TE02/A-TB350-TE02-ND/4329406), [Sparkfun PRT-08084](https://www.sparkfun.com/products/8084)).

Datasheet: [ASS_4091_CO.pdf](https://github.com/TinkerMill/tinkerAccess/files/4907787/ASS_4091_CO.pdf)

lttech-llc, 2020-08-24:
We decided in the design review to not implement this change for now

lttech-llc, 2020-09-08:
Not changed in PCB rev 1.4